ros/ros #78: Man pages

lepalom, 2015-02-03:
Please, could you write a man page for the following programs:
- binary-without-manpage usr/bin/rosmake
- binary-without-manpage usr/bin/roscreate-pkg
- binary-without-manpage usr/bin/rosboost-cfg
- binary-without-manpage usr/bin/rosunit
- binary-without-manpage usr/bin/rosrun
- binary-without-manpage usr/bin/rosclean

Although is not mandatory, it's a good practice in unix system, specially in debian and derivatives that every binary has its own manpage.

dirk-thomas, 2015-02-04:
All ROS documentation is captured in the wiki (wiki.ros.org) as well as in the documentation generated from the source code (docs.ros.org). It is very unlikely that we will have the time to write additional man pages especially with the continuous effort required to keep the content in sync.

lepalom, 2015-02-04:
With pcl we used help2man. Some --help messages are needed. help2man convert this output in a man page. With some small effort, this tools can have a --help complete (and maintainable) and after that, generate the man page is trivial.

dirk-thomas, 2015-02-04:
I don't think the maintainers do have the necessary bandwidth to work on this feature. If you would like to work on adding this to the ROS repositories that would be great.

LordFluffyPants, 2016-11-17:
I am willing to work on this problem, however I am unsure about how to go about submitting the man page files.

dirk-thomas, 2019-08-12:
Closing due to no activity.

closed 2019-08-12 by dirk-thomas