ros/ros #84: Cannot build jade roslib with OS X Yosemite

jmtatsch, 2015-06-01:
There is a indigo installation in /opt/ros/indigo, but its not sourced and I logged in again...
Can someone help?
---
cd /Users/tatsch/ros_jade_ws/build_isolated/roslib && /opt/ros/jade/env.sh cmake /Users/tatsch/ros_jade_ws/src/ros/roslib -DCATKIN_DEVEL_PREFIX=/Users/tatsch/ros_jade_ws/devel_isolated/roslib -DCMAKE_INSTALL_PREFIX=/opt/ros/jade -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'
-- Using CATKIN_DEVEL_PREFIX: /Users/tatsch/ros_jade_ws/devel_isolated/roslib
-- Using CMAKE_PREFIX_PATH: /opt/ros/jade;/Users/tatsch/ros_catkin_ws/devel;/opt/ros/indigo;/usr/local
-- This workspace overlays: /opt/ros/jade;/Users/tatsch/ros_catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python
-- Using default Python package layout
-- Using empy: /usr/local/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /Users/tatsch/ros_jade_ws/build_isolated/roslib/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.6.14
Traceback (most recent call last):
  File "/Users/tatsch/ros_jade_ws/build_isolated/roslib/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/Users/tatsch/ros_jade_ws/devel_isolated/roslib/env.sh')
  File "/opt/ros/jade/lib/python2.7/site-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
  File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ast.py", line 49, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ast.py", line 37, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    ROS_DISTRO was set to 'indigo' before. Please make sure that the environment does not mix paths from different distributions.
                 ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/jade/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/local/bin/python
  "/Users/tatsch/ros_jade_ws/build_isolated/roslib/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/jade/share/catkin/cmake/all.cmake:186 (safe_execute_process)
  /opt/ros/jade/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:3 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/tatsch/ros_jade_ws/build_isolated/roslib/CMakeFiles/CMakeOutput.log".

dirk-thomas, 2015-06-01:
Please post questions on answers.ros.org following the support guidelines (http://wiki.ros.org/Support#Guidelines_for_asking_a_question_.28Please_read_before_posting.29).

closed 2015-06-01 by dirk-thomas

wjwwood, 2015-06-01:
This error should be handled more gracefully, but I think you've still got the Indigo setup file sourced somehow.

I'd recommend asking something like this on answers.ros.org first, to make sure the problem is a bug. Make sure to tag with `catkin`.

jmtatsch, 2015-06-02:
Of course I checked rosanswers and applied the advice I found there before posting here. It just didn't mention there that 
`export ROS_WORKSPACE=~/ros_catkin_ws`
counts as sourcing a setup file...
Smooth building from there on :), thanks for your great work!