ros/ros #85: rosrun completion incorrect for files with spaces

trainman419, 2015-06-10:
When tab-completing a filename with spaces with rosrun, the space in the filename is not escaped properly.

For example:

```
touch 'foo bar'
rosrun roscpp_tutorials talker foo<TAB>
```

Completes to:
```
rosrun roscpp_tutorials talker foo bar
```

Instead, it should complete with the space escaped:
```
 rosrun roscpp_tutorials talker foo\ bar
```

(obviously the talker node doesn't take files as arguments, but the example serves to demonstrate how tab-completion is incorrect).

dirk-thomas, 2015-06-10:
The completion scripts are not designed to handle filenames with spaces (and I would expect other tools will also fall short in this cases).

If you would like to see support for it please consider contributing a PR for this. The maintainers of the package will likely not have the time to work on this enhancement.

de-vri-es, 2015-10-08:
I'm assuming this is for bash completion? Interestingly enough, the zsh completion does properly escape the completion candidate but then rosrun chokes on the space.

PR #94 (2015-10-13): rosrun: Allow spaces in command names and search paths.
This PR allows command names and search paths for `rosrun` to contain spaces. It mainly adds quotes around a bunch of variables and sets `IFS=$'\n' ` on two places to split only on newlines.

In the case of `$catkin_package_libexec_dr`, I turned it into an array (also by splitting the output with `IFS=$'\n'`, and renamed it to be plural) so the individual elements can contain spaces when passing it to `find`.

I tested with a test packages containing one script with a space in the name:
`rosrun test_package "test script"`

Without the PR, [... cut]
closed 2015-10-13 by dirk-thomas