ros/ros #90: [trivial] Correction in the wiki

WurmD, 2015-09-18:
at http://wiki.ros.org/action/login/ROS/Tutorials/NavigatingTheFilesystem

you can now change 
/opt/ros/hydro/share/roscpp
to
/opt/ros/indigo/share/roscpp

and
/opt/ros/indigo/base/install/share:/opt/ros/indigo/base/install/stacks
to
/opt/ros/indigo/base/install/share:/opt/ros/indigo/base/install/stacks

dirk-thomas, 2015-09-18:
As you have pointed out it is a trivial change. Since it is a wiki everyone can contribute to please go ahead and update the wiki page.

WurmD, 2015-09-18:
When I double click to edit, this is what appears instead of the page code:

##NavigatingTheFilesystem
####################################
##FILL ME IN
####################################
## links to any required tutorials
## note.0=[[ROS/Tutorials/InstallingandConfiguringROSEnvironment|Installing and Configuring Your ROS Environment]]
## descriptive title for the tutorial
## title = Navigating the ROS Filesystem
## multi-line description to be displayed in search
## description = This tutorial introduces ROS filesystem concepts, and covers using the roscd, rosls, and [[rospack]] commandline tools.
## the next tutorial description
## next =
## links to next tutorial
## next.0.link= [[ROS/Tutorials/CreatingPackage|Creating a ROS package]]
## what level user is this tutorial for
## level= BeginnerCategory
####################################
<<IncludeCSTemplate(TutorialCSHeaderTemplate)>>

<<Buildsystem()>>

{{{#!wiki buildsystem rosbuild
<<Include(ROS/Tutorials/rosbuild/NavigatingTheFilesystem)>>
}}}
{{{#!wiki buildsystem catkin
<<Include(ROS/Tutorials/catkin/NavigatingTheFilesystem)>>
}}}
Now that you can get around in ROS, let's [[http://wiki.ros.org/ROS/Tutorials/CreatingPackage|create a package]].

## AUTOGENERATED DO NOT DELETE
## TutorialCategory
## ROSTutorialCategory

dirk-thomas, 2015-09-18:
You need to escape a code block like this to prevent GitHub from interpreting the html inside.

If you look in the content of the page it conditionally includes two other pages. One of them is `ROS/Tutorials/catkin/NavigatingTheFilesystem`. If you go to that page you will find the content you are looking for to edit.

WurmD, 2015-09-18:
Done! ^_^

dirk-thomas, 2015-09-18:
Thanks!

http://wiki.ros.org/action/info/ROS/Tutorials/catkin/NavigatingTheFilesystem?action=diff&rev2=22&rev1=21

closed 2015-09-18 by dirk-thomas