ros/ros #96: d19b7cd breaks roscore

nxdefiant, 2015-10-18:
Hi,

since the last update my roscore is no longer working:

```
user@host:~> roscore                                                                                                                                                                                                                                                  0
... logging to /home/user/.ros/log/6bbc2caa-75c7-11e5-af44-0024d7aab6b8/roslaunch-host-6065.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://host:37425/
ros_comm version 1.11.15

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.15

NODES

auto-starting new master
process[master]: started with pid [6586]
ROS_MASTER_URI=http://host:11311/

setting /run_id to 6bbc2caa-75c7-11e5-af44-0024d7aab6b8
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/__init__.py", line 298, in main
    p.start()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 271, in start
    self.runner.launch()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/launch.py", line 646, in launch
    self._setup()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/launch.py", line 623, in _setup
    self._launch_core_nodes()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/launch.py", line 498, in _launch_core_nodes
    name, success = self.launch_node(node, core=True)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/launch.py", line 526, in launch_node
    process = create_node_process(self.run_id, node, master.uri)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/nodeprocess.py", line 131, in create_node_process
    args = create_local_process_args(node, machine)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/node_args.py", line 251, in create_local_process_args
    matches = roslib.packages.find_node(node.package, node.type, rospack=rospack)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslib/packages.py", line 400, in find_node
    return find_resource(pkg, node_type, filter_fn=_executable_filter, rospack=rospack)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslib/packages.py", line 497, in find_resource
    source_path_to_packages = rospack.get_custom_cache('source_path_to_packages', {})
AttributeError: 'RosPack' object has no attribute 'get_custom_cache'
[master] killing on exit
```

ros has been compiled from source.

dirk-thomas, 2015-10-18:
As the commit indicates (https://github.com/ros/ros/commit/d19b7cdb4b0d7e1b6f5388937599dfdf8b1f5464#diff-6fd5c0ae40f9eee7b3cfe7b55390ad5bR22) you need the current version of `rospkg`. Which version do you have installed?

nxdefiant, 2015-10-18:
Aah right I see, I would have expected rosinstall_generator to fetch the correct version for me, but rospkg is via pip. Nevermind.

closed 2015-10-18 by nxdefiant