ros/ros #120: gtest XML output lacks namespace

mikepurvis, 2016-09-09:
For repos with many packages, like ros_comm, the output as displayed by Jenkins can be a real jumble, with no clarity at all about where the different tests are coming from. The python tests are all listed somewhat by package in the base test results view, eg:

http://build.ros.org/job/Kdev__ros_comm__ubuntu_xenial_amd64/64/testReport/

But all the gtests are listed in `(root)` organized by the fixture/suite name, eg:

http://build.ros.org/job/Kdev__ros_comm__ubuntu_xenial_amd64/64/testReport/(root)/

Looking at one case in particular, the CheckMaster test is defined here:

https://github.com/ros/ros_comm/blob/kinetic-devel/test/test_roscpp/test/src/check_master.cpp#L49-L52

The result shows up here:

http://build.ros.org/job/Kdev__ros_comm__ubuntu_xenial_amd64/64/testReport/(root)/CheckMaster/checkMaster/

And the result XML looks like this:

```
<?xml version="1.0" encoding="UTF-8"?>
<testsuites tests="1" failures="0" disabled="0" errors="0" time="0.002" name="AllTests">
  <testsuite name="CheckMaster" tests="1" failures="0" disabled="0" errors="0" time="0.002">
    <testcase name="checkMaster" status="run" time="0.002" classname="CheckMaster" />
  </testsuite>
</testsuites>
```

IMO, the classname in this instance should be changed to be prefixed with the package name, so `test_roscpp.CheckMaster`. It appears that there's some preexisting [junit xml mangle logic here](https://github.com/ros/ros/blob/indigo-devel/tools/rosunit/src/rosunit/junitxml.py#L239), but it looks like it maybe runs as part of a summarizing script rather than as a post-step to every gtest run.

Thoughts on if there's some reasonable way forward here?

dirk-thomas, 2016-09-15:
I fully agree with you observation that the test results shown in Jenkins are not giving you any indicator where the tests are coming from.

I am not very keen to encode the package name into every gtest within that package. Simply because from a design point I don't think the test should know about it and the developer shouldn't need to repeat that information.

When you look at the generated JUnit / xUnit result files and where they are being generated you will notice that they are being generated in a subfolder named after the package. The problem of the Jenkins UI in my opinion is that it doesn't show you that file path. With that information you would know immediately from which package the test came from and even in which test group it is (e.g. a launch file or a specific gtest file).

It would be nice if the Jenkins plugin would be improved to actually show where the test result came from. That would help if you are looking at a specific result. But that wouldn't provide the nice hierarchy as for the Python tests (where you click on the package name first, and then the next level shows you the tests in there). So extending the mangling logic in the xml post-processing step to prefix gtest results with the package name are the only way to achieve that as far as I know.

dirk-thomas, 2019-08-12:
Closing due to no activity.

closed 2019-08-12 by dirk-thomas