ros/ros #128: roslib dependency to rospack not exported corretly to CMake?

Sarcasm, 2016-11-09:
ROS version: Indigo
Operating system: Ubuntu Trusty (14.04)
roslib package version: ros-indigo-roslib 1.11.13-0trusty-20160627-161857-0700

We notice a compilation issue in our environment.
The error happens in the case where we invoke CMake directly (not via catkin) and that the environment (setup.bash) is not sourced.
When building a node that depends on roslib (indirectly), we get a linker error:

```
ld: warning: librospack.so, needed by /opt/ros/indigo/lib/libroslib.so, not found (try using -rpath or -rpath-link)                                                                                    
/opt/ros/indigo/lib/libroslib.so: undefined reference to `rospack::ROSPack::run(std::string const&)'                                                                                     
collect2: error: ld returned 1 exit status                                                                          
```

Looking at the CMake config for roslib, the error makes sense:

    $ grep "set(libraries" /opt/ros/indigo/share/roslib/cmake/roslibConfig.cmake
    set(libraries "roslib")

Without going too deep in the roslib package I would expect to have:

    set(libraries "roslib;rospack")

Or even:

    set(libraries "rospack")

I wanted to propose a patch to add the dependency to the CMakeLists.txt and package.xml, however the information is already present:

- [indigo-devel@core/roslib/CMakeLists.txt#L3](https://github.com/ros/ros/blob/7bb3ba7a9b3a0102ba0a93efe28fcb03567d81ef/core/roslib/CMakeLists.txt#L3)
- [indigo-devel@core/roslib/package.xml#L19](https://github.com/ros/ros/blob/7bb3ba7a9b3a0102ba0a93efe28fcb03567d81ef/core/roslib/package.xml#L19)

Any idea why the roslibConfig.cmake does not list the dependencies of the CMakeLists.txt/package.xml?
I do not have a deep understanding of how the whole build is working so I may have made a few wrong assumptions.

dirk-thomas, 2016-11-09:
Can you please try to add `CATKIN_DEPENDS` to the `catkin_package(...)` call and check if that resolves the problem for you.

Sarcasm, 2016-11-10:
I had difficulty to check because I do not build  using the Ubuntu built package.
I now realize my fix to change the `libraries` variable is wrong and what needs to be updated is `depends`.

With your command `depends` is set properly and the build is working fine.

I have create a PR #129 against `kinetic-devel`.

closed 2016-11-10 by Sarcasm