ros/ros #142: Failure processing 'collada_parser' while building catkin workspace. ROS indigo on Rasberry Pi Jessie.

EscontrelaAlejandro, 2017-03-25:
For the past couple of days I have been following [This guide](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Indigo%20on%20Raspberry%20Pi) to install ROS indigo on my Raspberry Pi 2 with debian Jessie. I followed the guide without encountering any difficulties... until I started building the Catkin workspace. While processing the 87th package, it threw me an error saying:
`Failed to process package 'collada_parser'`

I have looked around multiple forums and tried some potential solutions including
`ldconfig -v`
But none seem to work.

There is one part of the guide which I didn't follow which was to [patch collada_urdf](https://github.com/ros/robot_model/issues/12). I'm not sure if this is the cause of this error since I am getting a failure with 'collada_parser' and not 'collada_urdf'.

The output pertaining to the collada_parser failure is as follows:

/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:1168:47: error: ‘class urdf::Link’ has no member named ‘visual’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:1169:18: error: ‘class urdf::Link’ has no member named ‘collision’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:1169:47: error: ‘class urdf::Link’ has no member named ‘visual’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp: In member function ‘void urdf::ColladaModelReader::_ExtractRobotAttachedActuators(ColladaDOM150::domArticulated_systemRef)’:
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2039:40: error: ‘class urdf::Joint’ has no member named ‘limits’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2040:37: error: ‘class urdf::Joint’ has no member named ‘limits’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2041:29: error: ‘class urdf::Joint’ has no member named ‘limits’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp: In member function ‘boost::shared_ptr<urdf::Joint> urdf::ColladaModelReader::_getJointFromRef(xsToken, daeElementRef)’:
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2434:29: error: request for member ‘find’ in ‘((urdf::ColladaModelReader*)this)->urdf::ColladaModelReader::_model.boost::shared_ptr<T>::operator-><urdf::ModelInterface>()->urdf::ModelInterface::joints_’, which is of non-class type ‘int’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2434:59: error: request for member ‘end’ in ‘((urdf::ColladaModelReader*)this)->urdf::ColladaModelReader::_model.boost::shared_ptr<T>::operator-><urdf::ModelInterface>()->urdf::ModelInterface::joints_’, which is of non-class type ‘int’
/home/pi/ros_catkin_ws/src/robot_model/collada_parser/src/collada_parser.cpp:2437:38: error: request for member ‘find’ in ‘((urdf::ColladaModelReader*)this)->urdf::ColladaModelReader::_model.boost::shared_ptr<T>::operator-><urdf::ModelInterface>()->urdf::ModelInterface::joints_’, which is of non-class type ‘int’
CMakeFiles/collada_parser.dir/build.make:62: recipe for target 'CMakeFiles/collada_parser.dir/src/collada_parser.cpp.o' failed
make[2]: *** [CMakeFiles/collada_parser.dir/src/collada_parser.cpp.o] Error 1
CMakeFiles/Makefile2:387: recipe for target 'CMakeFiles/collada_parser.dir/all' failed
make[1]: *** [CMakeFiles/collada_parser.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'collada_parser': 
  Command '['/opt/ros/indigo/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/pi/ros_catkin_ws/build_isolated/collada_parser && /opt/ros/indigo/env.sh make -j4 -l4

Command failed, exiting.

Any help would be greatly appreciated, thank you.

dirk-thomas, 2017-03-25:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2017-03-25 by dirk-thomas

EscontrelaAlejandro, 2017-03-25:
http://answers.ros.org/question/257852/failure-processing-collada_parser-while-building-catkin-workspace-ros-indigo-on-rasberry-pi-jessie/