ros/ros #145: roscp doesn't work for installed files not in a subfolder of the manifest

dirk-thomas, 2017-05-11:
E.g. `roscp roscpp_tutorials TwoInts.h .` fails since it only considers files under `/opt/ros/lunar/share/roscpp_tutorials`.

mikaelarguedas, 2017-05-11:
Applies to `rosed` command as well

AditiPawaskar, 2017-08-08:
I want to try and solve this bug. Could anyone please guide me how to go about it?

dirk-thomas, 2017-08-14:
The files are installed in an FHS compliant location. The manifest is located in `share/<pkgname>`. The `roscp` and `rosed` commands both use [_roscmd](https://github.com/ros/ros/blob/782d333f26f6f6f020412b3f56669ecace690d81/tools/rosbash/rosbash#L233) which only looks in a few possible locations. In order to work for other locations the function need to be extended or a new function being written which finds files in other locations.

PREM1980, 2018-02-22:
@dirk-thomas Planning to submit a PR on this issue. Want to make sure no one else is working on it.

dirk-thomas, 2018-02-22:
> Want to make sure no one else is working on it.

Nobody working on this afaik. Thank you, looking forward for your pull request.

PREM1980, 2018-02-22:
@dirk-thomas : Thanks.

I tried to execute the below command after installing lunar but get the following error - 

m-ubu1604:/opt/ros/lunar/share/roscpp_tutorials$ roscp roscpp_tutorials TwoInts.h .
That file does not exist in that package.
cp: missing destination file operand after '.'
Try 'cp --help' for more information.

What is the expected result?  or can I know what is the  problem?

dirk-thomas, 2018-02-22:
> That file does not exist in that package.

This error message indicates that the script didn't find the source file.

> cp: missing destination file operand after '.'
> Try 'cp --help' for more information.

This is probably a follow up error because the source was not found the destination `.` is the only argument to `cp` which is insufficient.

> What is the expected result? or can I know what is the problem?

The expected behavior is that the file file `TwoInts.h` in the package `roscpp_tutorials` is copied to the destination `.`. You will have to debug the logic in the `roscp` function to figure what went wrong and how to fix it.

PREM1980, 2018-02-22:
ok, I don't see the TwoInts.h within the roscpp_tutorials directory..

m-ubu1604:/opt/ros/lunar/share/roscpp_tutorials$ tree
.
├── cmake
│   ├── roscpp_tutorialsConfig.cmake
│   ├── roscpp_tutorialsConfig-version.cmake
│   ├── roscpp_tutorials-msg-extras.cmake
│   └── roscpp_tutorials-msg-paths.cmake
├── launch
│   └── talker_listener.launch
├── package.xml
└── srv
    └── TwoInts.srv

Shouldn't the TwoInts.h should be located within the roscpp_tutorials directory. Currently, its installed by default in the following directory - 

m-ubu1604:/opt/ros/lunar/share/roscpp_tutorials$

dirk-thomas, 2018-02-22:
The header files can be found in the `include` directory of the package: e.g. `/opt/ros/lunar/include/roscpp_tutorials/TwoInts.h`

PREM1980, 2018-02-23:
I'm trying to debug the _roscmd that you have mentioned above. Its a bash function, If I add some debug stmts to this file its need to built again, correct?

I see these couple of make files under the rosbash folder but from which folder do I run the cmake command?

rosbashConfig.cmake
rosbashConfig-version.cmake

I'm more a python developer and kind of new to ROS, so trying to get my feet wet.

PREM1980, 2018-02-23:
Need some inputs.

Currently, in the _roscmd function the pkdir is found by calling the _ros_package_find function.

pkg_dir = /opt/ros/lunar/share/roscpp_tutorials

What would be the next level of search? 

1. It will go against /include directories for any packages that is installed(ie; one step before). 

2. Can I safely assume that all the services file created will be in the /include directories for all custom packages as well?

dirk-thomas, 2018-02-26:
> It will go against /include directories for any packages that is installed(ie; one step before).

I don't understand what your question here is?

> Can I safely assume that all the services file created will be in the /include directories for all custom packages as well?

Yes, if they follow the convention.

That is in general only true if the package was installed. If it was only build and is being used from the `devel` space this will not be the case. (And there is no good heuristic I am aware to conclude about the location of the headers in that case.)

PREM1980, 2018-02-26:
1. It will go against /include directories for any packages that is installed(ie; one step before).
        **What I meant here is - Should my new bash script should search the /include directories for the given package(roscpp_tutorial) and copy the files.**

To conclude, after searching the /share directories the new script will check for any files in the /include directories and copy those files. Will this suffice to fix this issue?

dirk-thomas, 2018-02-26:
> Will this suffice to fix this issue?

That will work when packages install their headers in a subdirectory named after the package. (And as mentioned before only when being installed - not in the devel space.)

PREM1980, 2018-02-27:
1. ok, so, how can I determine if the subdirectory is named after the package or not? What would be the correct approach to find it?
2. Also, how do we handle components on the devel space? Should I just ignore it for now?

dirk-thomas, 2018-03-05:
> how can I determine if the subdirectory is named after the package or not?

I don't think there is another approach beside hoping the package is following conventions.

> how do we handle components on the devel space? Should I just ignore it for now?

In order to handle the devel space there would need to be a per-package information where the corresponding source space is. That currently doesn't exist and I don't have a good proposal how to provide it. You could ignore it for now - I am just not sure what to do when the command doesn't work in that currently common case.

PREM1980, 2018-03-12:
The location of the package directory is determined by the rospack package.

ex:- rospack find

Currently, there are few options for the rospack command.

libs-only-L - finds the /lib directory of the package.

Should I add one more option(to rospack) to determine the /include directory path and the use that output to copy the files?

closed 2025-05-31 by sloretz