ros/ros #146: Floating point exception (core dumped) while sourcing setup.bash file

yoshimalucky, 2017-05-22:
I got an error "Floating point exception (core dumped)" while sourcing /opt/ros/indigo/setup.bash file.

It looks like not affecting the system, but I'm curious why this error message occurs.

Thanks.

dirk-thomas, 2017-05-22:
The setup file itself doesn't run any binary program. It could be any of the environment hooks. Can you please edit the file `/opt/ros/indigo/setup.sh` and find the line containing`. "$_envfile"`. Then insert a line above outputting the environment hook filename: `echo "envhook: $_envfile"`. Then try sourcing the setup file again. Maybe the line printed before the crash will allow to narrow down the problem...

jbroadus, 2017-05-22:
Since your error message indicated that a core was dumped, you can look at
the corefile with gdb. Look at /proc/sys/kernel/core_pattern for the core
dump path.

dirk-thomas, 2017-07-21:
I will close this ticket due to no response. Please feel free to comment and the ticket can be reopened if necessary.

closed 2017-07-21 by dirk-thomas

deepakgangwar, 2017-11-14:
I am also facing the same problem while sourcing setup.zsh but the cause of error should be same as bot zsh and bash scripts source the setup.sh file. I did the step as mentioned in the [comment](https://github.com/ros/ros/issues/146#issuecomment-303233954l). The output is following -
`
envhook: /opt/ros/indigo/etc/catkin/profile.d/00.ocl-lua.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/00.rtt.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/00.typelib.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/00.utilrb.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/10.kdl_typekit.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/10.rfsm.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/10.ros.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/10.rosbuild.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/10.roslaunch.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/15.rosjava.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.roch-navigation.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.roch.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.rocon-ninjablock-bridge.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.rocon-rtsp-camera-relay.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.rocon-smartthings-bridge.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.turtlebot-navigation.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.turtlebot-stage.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.turtlebot-stdr.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/25.turtlebot.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/50.husky_description.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/50.husky_find_mag_config.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/50.roch_description.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.euslisp.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.jsk_tools.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.openrave.sh
zsh: floating point exception (core dumped)  
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.roslisp.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.rtmlaunch.sh
envhook: /opt/ros/indigo/etc/catkin/profile.d/15.rocon_app.zsh
envhook: /opt/ros/indigo/etc/catkin/profile.d/15.rocon_launch.zsh
envhook: /opt/ros/indigo/etc/catkin/profile.d/15.rocon_test.zsh
envhook: /opt/ros/indigo/etc/catkin/profile.d/15.rosbash.zsh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.config_switch.zsh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.jsk_tools-completion.zsh
envhook: /opt/ros/indigo/etc/catkin/profile.d/99.jsk_tools.zsh

`

yoshimalucky, 2017-11-14:
@deepakgangwar 

I found that "envhook: /opt/ros/indigo/etc/catkin/profile.d/99.openrave.sh" cause the problem, so I uninstalled openrave. If you really need it, report this issue to:
https://github.com/jsk-ros-pkg/openrave_planning

deepakgangwar, 2017-11-14:
@yoshimalucky Thanks, this solved the problem.

dirk-thomas, 2017-11-14:
Thank you for commenting with the additional findings. Since the problem is not in this repo I will close this ticket. I would highly recommend to create a follow up ticket in the referenced openrave repository in order to get this fixed. Please comment here with a link to the created ticket for future reference.