ros/ros #148: rosunit invalid_chars

Alessandro-Barbieri, 2017-06-24:
When installing from source

```
running build
running build_py
running build_scripts
running install
running install_lib
byte-compiling /opt/ros/lunar/lib64/python3.5/site-packages/rosunit/junitxml.py to junitxml.cpython-35.pyc
byte-compiling /opt/ros/lunar/lib64/python3.5/site-packages/rosunit/xmlrunner.py to xmlrunner.cpython-35.pyc
running install_scripts
changing mode of /opt/ros/lunar/bin/rosunit to 755
running install_egg_info
Removing /opt/ros/lunar/lib64/python3.5/site-packages/rosunit-1.14.0-py3.5.egg-info
Writing /opt/ros/lunar/lib64/python3.5/site-packages/rosunit-1.14.0-py3.5.egg-info
  File "/opt/ros/lunar/lib64/python3.5/site-packages/rosunit/junitxml.py", line 59
    invalid_chars = re.compile(ur'[^\x09\x0A\x0D\x20-\x7E\x85\xA0-\xFF\u0100-\uD7FF\uE000-\uFDCF\uFDE0-\uFFFD]')
                                                                                                              ^
SyntaxError: invalid syntax

  File "/opt/ros/lunar/lib64/python3.5/site-packages/rosunit/xmlrunner.py", line 162
    invalid_chars = re.compile(ur'[^\x09\x0A\x0D\x20-\x7E\x85\xA0-\xFF\u0100-\uD7FF\uE000-\uFDCF\uFDE0-\uFFFD]')
                                                                                                              ^
SyntaxError: invalid syntax

-- Installing: /opt/ros/lunar/share/rosunit/scripts/check_test_ran.py
-- Installing: /opt/ros/lunar/share/rosunit/scripts/clean_junit_xml.py
-- Installing: /opt/ros/lunar/share/rosunit/scripts/pycoverage_to_html.py
-- Installing: /opt/ros/lunar/share/rosunit/scripts/summarize_results.py
-- Installing: /opt/ros/lunar/share/rosunit/scripts/test_results_dir.py
<== Finished processing package [74 of 241]: 'rosunit'

```

tfoote, 2017-06-26:
Please provide enough information to reproduce this. I see you're using python3.5 anything else different?

dirk-thomas, 2017-06-26:
The syntax for a string `ur'value'` is invalid in Python 3. No need for more information.

Alessandro-Barbieri, 2017-06-26:
Feel free to close as ROS default python is 2

dirk-thomas, 2017-06-26:
We will keep it open. While we don't claim to support Python 3 we still aim to make it work since several platforms are using it nowadays.

Any PR for this would likely speed up the fixing process and would be highly appreciated :wink:

Alessandro-Barbieri, 2017-07-02:
@aballier can you drop all the python3 support for this package in the ebuild?

aballier, 2017-07-02:
there are no patch in the ebuild;
it installs fine, allows some usage of rosunit to work with python3 but rosunit fails its testsuite with python3 and causes some test failures on other packages with python3 too, this issue is the core of the problem I think

dirk-thomas, 2017-07-06:
@Alessandro-Barbieri Can you please try the proposed patch from #150 and comment if that resolves the problem for you with Python 3.

Alessandro-Barbieri, 2017-07-07:
@dirk-thomas I need guidance on how to do this.
I have lunar installed with python2.7, I can start the installation with python3.5 but how can I test it while is installing for the first time?

dirk-thomas, 2017-07-07:
@Alessandro-Barbieri You can't use or source your installed ROS packages for this. Instead start with an empty workspace and clone this branch as well as all the repos / packages for recursive dependencies. When you invoke the from source build of that workspace you pass the Python 3 interpreter as a CMake argument: `-DPYTHON_EXECUTABLE=/path/to/python3.5`

For example something like this to only test the minimal set of packages:
* `mkdir -p /tmp/ws/src && cd /tmp/ws`
* `rosinstall_generator --rosdistro lunar rosunit --deps > list.repos`
* `vcs import src < list.repos`
* `./src/catkin/bin/catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3`
* `./src/catkin/bin/catkin_make run_tests`

This will fail with the reported `SyntaxError`. If you now apply the patch and run the tests again that problem should go away (there will be other test failures with Python 3 though).

PR #150 (2017-07-21): fix syntax of unicode raw string in Python 3
Fixes #148.
closed 2017-07-21 by dirk-thomas