ros/ros #149: How to make ROS and anaconda work on Ubuntu at the same time

ramadawn, 2017-07-03:
Typically I have found that ROS will interfere with an installation of anaconda on Ubuntu (or vice versa).

The way to resolve this problem is to comment out the line:

`export PATH=/usr/bin/anaconda/bin:$PATH' in your BASHRC file.

Then when you need to run anaconda-navigator:

1.open a terminal

2. Type: export PATH=/usr/bin/anaconda/bin:$PATH

3. and then activate anaconda navigator by typing:
   ` anaconda-navigator

This will allow you to use ROS and anaconda on the same install.

dirk-thomas, 2017-07-03:
This is great information for anaconda users. This issue is probably not the best location for others to find out about it. Please consider to add the information to the ROS wiki (where ever you think it is most appropriate) and post a link here.

dirk-thomas, 2017-07-21:
@ramadawn As mentioned in the previous comment the issue tracker in this repo is used to track bugs, feature requests, etc. But this ticket is not such an action item. Therefore I will close it for now. Please consider contributing to the wiki if you think this information should be more visible to users.

closed 2017-07-21 by dirk-thomas

ramadawn, 2017-07-21:
Thanks I already updated the ROS wiki to describe a solution to the problem.

http://wiki.ros.org/IDEs

StefanFabian, 2018-05-04:
I wrote a script that handles the ROS + Anaconda conflict a tad more convenient which can be found here:
https://gist.github.com/StefanFabian/17fa715e783cd2be6a32cd5bbb98acd9

It works by creating aliases for all executables provided by anaconda that aren't present on the system already (so I'm not sure if it would work with a global install in /usr/bin). Whenever one of these executables is called, the script adds Anaconda to the path, removes ROS from the PYTHONPATH, executes the command and afterwards removes anaconda from the path again. This, of course, also works with `source activate` which will keep Anaconda in the path until `source deactivate` is executed.

axn337, 2018-11-05:
Thank you ramadawn,

I am still getting an error when I catkin_make, even though I removed this line from .bashrc 
export PATH=/opt/anaconda/anaconda3/bin:$PATH

Note I have installed anaconda globally in /opt/ for all the users. 
Am I missing something?

Thank you

warnerjon12, 2018-11-17:
I have my PS3 set to show which Anaconda environment I am in. By default it shows 'base'. If I just run 'conda deactivate', I can check that the anaconda prefix is indeed removed from PATH. This may not work with older versions of Anaconda.

Ankur-Deka, 2020-03-25:
https://github.com/Ankur-Deka/PyTorch-with-ROS-Installation-Guide

rickstaa, 2020-10-13:
For future reference, I wrote a small installable wrapper based on the code of @StefanFabian. It can be found [here](https://github.com/rickstaa/.ros_conda_wrapper).

Tobias-Fischer, 2021-02-19:
It is now also possible to install ROS packages directly (natively) in a conda environment: https://github.com/RoboStack/ros-noetic & https://medium.com/robostack/cross-platform-conda-packages-for-ros-fa1974fd1de3

rickstaa, 2021-02-23:
@Tobias-Fischer Thanks for updating me on the fact that you fully ported ROS-NOETIC. I will add it as the default in my wrapper might people want to use it alongside the desktop ROS version.

Tobias-Fischer, 2021-02-23:
Hi @rickstaa! That would be interesting indeed. However with the Robostack ROS conda packages, there should not be need for a wrapper anymore. The only thing that needs doing if a system ROS is also installed is `unset PYTHONPATH`

rickstaa, 2021-02-26:
@Tobias-Fischer Thanks for the updates. That's amazing to hear. I will test it out later this month and add a deprecation warning to the wrapper pointing to your packages.