ros/ros #156: move roslib's env-hooks into dedicated package

ipa-mdl, 2017-11-08:
`rospack`, `rospkg`, etc. rely on `ROS_PACKAGE_PATH` to crawl all workspaces.
At the moment `roslib` registers an env-hook for catkin that prefills the `ROS_PACKAGE_PATH` based on `CMAKE_PREFIX_PATH`.

So if `roslib` is not present in a sytem (e.g. the `ros_buildfarm` or `industral_ci` images) `rospack` will not be able to find anything.

Since `roslib` already depends on `rospack`, the `exec_depend` cannot added to the `rospack`.

That's why I propose to create a new package (`rosenvhooks`?) that just contains the hooks.
Afterwards `rospack` and `roslib` and any other package that reads  `ROS_PACKAGE_PATH` can depend on it.

dirk-thomas, 2018-01-02:
I just created ros-infrastructure/rep#151 to clarify the goal of having a new separate package for the new environment variables defined in REP 149. Other environment variables can then be moved to this new package too.

dirk-thomas, 2018-01-26:
The new package [ros_environment](https://github.com/ros/ros_environment) has been created and released into Kinetic, Lunar and Melodic.

I will keep this ticket open until `roslib` has been updated to use the new package and removed its own environment hooks.

PR #166 (2018-01-26): replace env hooks with a dependency on ros_environment
Depending on the new package [ros_environment](https://github.com/ros/ros_environment) which has distro specific branches allows this package to consolidate the `kinetic-devel` and `lunar-devel` branches as well as use the same branch also for the upcoming `melodic` release.

Fixes #156.
closed 2018-01-26 by dirk-thomas