ros/ros #158: Unit test failures in python 3

paulbovbel, 2017-12-14:
Running the unit tests under nosetests3 reveals some inconsistencies in byte/string handling:

```
nosetests3 test_xmlrunner.py 
EEEEEEEE
======================================================================
ERROR: Regression test: Check whether a test run with a erroneous test
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pbovbel/locus_dev/src/ros/tools/rosunit/test/test_xmlrunner.py", line 87, in test_error
    self._try_test_run(TestTest, """<testsuite errors="1" failures="0" name="unittest.suite.TestSuite" tests="1" time="0.000"><testcase classname="test.test_xmlrunner.TestTest" name="test_foo" time="0.000"><error type="IndexError">Foobar</error></testcase><system-out>&lt;![CDATA[\n\n]]&gt;</system-out><system-err>&lt;![CDATA[\n\n]]&gt;</system-err></testsuite>""")
  File "/home/pbovbel/locus_dev/src/ros/tools/rosunit/test/test_xmlrunner.py", line 37, in _try_test_run
    runner.run(unittest.makeSuite(test_class))
  File "/home/pbovbel/locus_dev/src/ros/tools/rosunit/src/rosunit/xmlrunner.py", line 275, in run
    result.print_report(stream, time_taken, out_s, err_s)
  File "/home/pbovbel/locus_dev/src/ros/tools/rosunit/src/rosunit/xmlrunner.py", line 202, in print_report
    stream.write(ET.tostring(self.xml(time_taken, out, err).getroot(), encoding='utf-8', method='xml'))
TypeError: string argument expected, got 'bytes'
...
```

and

```
nosetests3 test_junitxml.py 
FFF.......
======================================================================
FAIL: testGenerateError (test.test_junitxml.XmlResultTestGeneration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pbovbel/locus_dev/src/ros/tools/rosunit/test/test_junitxml.py", line 109, in testGenerateError
    ]]&gt;</error>''', error_str)
AssertionError: '<error type="error_type">&lt;![CDATA[\nerror_text\n]]&gt;</error>' != b'<error type="error_type">&lt;![CDATA[\nerror_text\n]]&gt;</error>'
...
```

felixdivo, 2018-01-24:
I got the same after executing the tests with `catkin_make run_tests -DPYTHON_EXECUTABLE=/usr/bin/python3.5 -DPYTHON_INCLUDE_DIR=/usr/include/python3.5 -DPYTHON_LIBRARY=/usr/lib/libpython3.5.so`:

```
Traceback (most recent call last):
  File "/home/felix/sailing/bord-pc/src/first_package/tests-node/test-correct-response.py", line 19, in <module>
    rostest.rosrun('first_package', 'my_bare_bones_test', TestBareBones)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/__init__.py", line 146, in rosrun
    result = rosunit.create_xml_runner(package, test_name, result_file).run(suite)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosunit/xmlrunner.py", line 275, in run
    result.print_report(stream, time_taken, out_s, err_s)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosunit/xmlrunner.py", line 202, in print_report
    stream.write(ET.tostring(self.xml(time_taken, out, err).getroot(), encoding='utf-8', method='xml'))
TypeError: write() argument must be str, not bytes
```

dirk-thomas, 2018-01-24:
Please provide a [sscce](http://sscce.org/) to reproduce the problem.

paulbovbel, 2018-01-24:
The following should work:

```
sudo apt-get install python3-nose
roscd rosunit
nosetests3 test/test_xmlrunner.py
```

dirk-thomas, 2018-01-24:
Thank you for providing the steps. I can reproduce the problem with them.

Please consider to provide a pull request to address the problem since it is unclear when the maintainers will find time to work on Python 3 support.

felixdivo, 2018-01-24:
I will have a look at it.

ablakey, 2018-08-21:
For a bit more context, the unit tests continue to all pass for Python 2. They continue to fail for Python 3 because of a slight change in the output:

```
- <testsuite errors="0" failures="0" name="unittest.suite.TestSuite" tests="1" time="0.000"><testcase classname="test.test_xmlrunner.TestTest" name="test_foo" time="0.000" /><system-out>&lt;![CDATA[
?                                                                                                                                    ^^^^^^^^^
+ <testsuite errors="0" failures="0" name="unittest.suite.TestSuite" tests="1" time="0.000"><testcase classname="test.test_xmlrunner.XMLTestRunnerTest.test_stdout_capture.&lt;locals&gt;.TestTest" name="test_foo" time="0.000" /><system-out>&lt;![CDATA[
?                                                                                                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

I could make all the tests pass by providing both the Python2 and Python3 expected output if that helps. I don't think we run these tests against Python3 though, maybe?

I dug into this because it was blocking work I had when running tests in Python 3. So I can personally say that it addresses this very problem (I'm using the same library that @paulbovbel used to originally hit this bug and create this ticket). But I agree that additional parties chiming in would be helpful!