ros/ros #161: Bash completion filters out to many files

nalt, 2018-01-03:
The function _roscomplete_find for tab completion in rosbash filters out hidden files and folders using `"! -regex ".*/[.].*"`.

If a ".." ends up in the package path - which happens when the user manually adds a relative package path, and also for some workspace configurations using catkin, those paths are filtered out as well. As a result, tab completion finds nothing in the affected packages.

This problem could be avoided by changing the regex to `".*/[.][^.].*"`

dirk-thomas, 2018-01-03:
Please provide a reproducible example.

nalt, 2018-01-03:
Set `ROS_PACKAGE_PATH=/home/user/ros/ws/../mypack:/opt/ros/kinetic/share`. Now type `rosrun mypack <tab>`. It does not list anything. The path with .. is filtered out by `"! -regex ".*/[.].*"` in: https://github.com/ros/ros/blob/ba451d78b1482101b239328beae5b8e809b1e272/tools/rosbash/rosbash#L459

One can question whether these `..` should be there. Still, this is a valid path.

Now, set `ROS_PACKAGE_PATH=/home/user/ros/mypack:/opt/ros/kinetic/share`. Now type `rosrun mypack <tab>`. It lists the executables, as expected.

dirk-thomas, 2018-01-03:
Thank you for the example. Please consider to provide a pull request to support this case.

tfoote, 2018-01-03:
Another approach would be to resolve the path before running the regex. There are a few different potential approaches here: https://stackoverflow.com/questions/3915040/bash-fish-command-to-print-absolute-path-to-a-file

nalt, 2018-01-03:
Do you have a preferred solution? All these seem to have their disadvantages, e.g. they work only for one path. `realpath` would do the job, but it may print errors.

dirk-thomas, 2018-01-04:
I am agnostic to how this is being addressed (as long as it makes your use case work without any drawbacks).

dirk-thomas, 2018-01-30:
Addressed in #162.

closed 2018-01-30 by dirk-thomas