ros/ros #191: 15 undefined names

cclauss, 2018-08-24:
[flake8](http://flake8.pycqa.org) testing of https://github.com/ros/ros on Python 3.6.3

$ __flake8 . --count --select=E901,E999,F821,F822,F823 --show-source --statistics__
```
./core/roslib/src/roslib/message.py:83:17: F821 undefined name 'reload'
                reload(pypkg)
                ^
./core/roslib/src/roslib/names.py:62:30: F821 undefined name 'basestring'
        return isinstance(s, basestring) #Python 2.x
                             ^
./core/roslib/src/roslib/packages.py:224:20: F821 undefined name 'error'
            except error:
                   ^
./core/roslib/src/roslib/scriptutil.py:85:16: F821 undefined name 'ns_join'
        return ns_join(roslib.names.make_caller_id(script_name), name[1:])
               ^
./core/roslib/src/roslib/stacks.py:235:23: F821 undefined name 'ReleaseException'
                raise ReleaseException("couldn't find version number in CMakeLists.txt:\n\n%s"%l)
                      ^
./tools/rosmake/src/rosmake/engine.py:307:37: F821 undefined name 'argument'
        return len(list(self.result[argument].keys())) #py3k
                                    ^
./tools/rosunit/src/rosunit/baretest.py:261:27: F821 undefined name 'RLException'
                    raise RLException("unable to create directory for log file [%s].\nPlease check permissions."%log_dir)
                          ^
./tools/rosunit/src/rosunit/baretest.py:263:27: F821 undefined name 'RLException'
                    raise RLException("unable to create directory for log file [%s]: %s"%(log_dir, e.strerror))
                          ^
./tools/rosunit/src/rosunit/baretest.py:280:16: F821 undefined name 'is_child_mode'
            if is_child_mode():
               ^
./tools/rosunit/src/rosunit/baretest.py:287:25: F821 undefined name '_cleanup_remappings'
            self.args = _cleanup_remappings(self.args, '__log:=')
                        ^
./tools/rosunit/src/rosunit/baretest.py:321:23: F821 undefined name 'get_ros_root'
                cwd = get_ros_root()
                      ^
./tools/rosunit/src/rosunit/baretest.py:343:73: F821 undefined name 'msg'
executable permission. This is often caused by a bad launch-prefix."""%(msg, ' '.join(self.args)))
                                                                        ^
./tools/rosunit/src/rosunit/baretest.py:345:101: F821 undefined name 'msg'
                    raise pmon.FatalProcessLaunch("unable to launch [%s]: %s"%(' '.join(self.args), msg))
                                                                                                    ^
./tools/rosunit/src/rosunit/pmon.py:324:33: F821 undefined name 'basestring'
        if not isinstance(name, basestring):
                                ^
./tools/rosunit/src/rosunit/xmlrunner.py:246:22: F821 undefined name 'file'
            stream = file(os.path.join(self._path, filename), "w")
                     ^
15    F821 undefined name 'reload'
15
```

dirk-thomas, 2018-08-24:
Please consider to provide pull requests to address these.

dirk-thomas, 2019-08-12:
Closing due to inactivity.

closed 2019-08-12 by dirk-thomas

cclauss, 2019-08-12:
Just for completeness flake8 on Python 3 still finds 15 _undefined names_ in this codebase.