ros/ros #192: rosbash: "find: paths must precede expression" on auto-complete attempt of rosmsg

gavanderhoorn, 2018-09-04:
To reproduce:

```
docker run -it --rm ros:melodic
source /opt/ros/melodic/setup.bash
roscd sensor_msgs/msg
rosmsg show sen<tab>
```

error returned then is:

```
find: paths must precede expression: `CameraInfo.msg'
find: possible unquoted pattern after predicate `-name'?
```

I've tried to understand what is going on in the auto-complete hook for `rosmsg`, but I've never had the pleasure of working with `compgen` so it's all dark magic to me. It seems like the problem *could be* somewhere around:

https://github.com/ros/ros/blob/fd860146ff14efc425ef020ce570b3ba1e950772/tools/rosbash/rosbash#L902-L905

gavanderhoorn, 2018-09-04:
I have a hunch that something like [this](https://stackoverflow.com/questions/27922449) is happening.

jexner, 2019-05-16:
@gavanderhoorn Yes, that was the actual issue. #218 provides a fix.

dirk-thomas, 2019-08-12:
Addressed by #218.

closed 2019-08-12 by dirk-thomas