ros/ros #195: Please add a license to this repo

mbrukman, 2019-01-22:
Could you please add an explicit `LICENSE` file to the repo so that it's clear under what terms the content is provided, and under what terms user contributions are licensed?

Per [GitHub docs on licensing](https://help.github.com/articles/licensing-a-repository/#choosing-the-right-license):

> [...] without a license, the default copyright laws apply, meaning that you retain all rights to your source code and no one may reproduce, distribute, or create derivative works from your work. If you're creating an open source project, we strongly encourage you to include an open source license.

Thanks!

dirk-thomas, 2019-01-22:
Each package in this repository specifies it's license in the `package.xml` file already.

mbrukman, 2019-01-22:
Thanks for the info! Are they all the same or different? If they're all the same, adding a `LICENSE` file it makes it easier for viewers on GitHub to see what the license directly from the repo overview, thanks to this feature: https://github.blog/2016-09-21-license-now-displayed-on-repository-overview/ so this would be a welcome addition to clarify the licensing.

dirk-thomas, 2019-01-22:
If someone could check all of them and post a summary here that would be great.

gavanderhoorn, 2019-06-04:
A quick check of all `package.xml`s shows:

```
mk: bsd
ros: bsd
rosbash: bsd
rosboost_cfg: bsd
rosbuild: bsd
rosclean: bsd
roscreate: bsd
roslang: bsd
roslib: bsd
rosmake: bsd
rosunit: bsd
```

amilcarlucas, 2019-06-20:
I also need this. But I can not do a PR for legal reasons. :(

amilcarlucas, 2019-09-21:
Fixed by #222 . Can be closed now.

closed 2019-09-21 by dirk-thomas