ros/ros #201: rosbag: why use the Crypto package ?

CdeMills, 2019-01-25:


dirk-thomas, 2019-01-25:
This ticket seems to be in the wrong place. Nothing in this repository uses `Crypto`. I assume you are referring to [ros_comm](https://github.com/ros/ros_comm/)?

> after installing and started the tutorials of "Melodic Morenia", I found the command "rostopic" failing at bag.py:
> from Crypto import Random

That sounds weird since the package declares a dependency on `python-crypto` which should be installed. So this shouldn't fail. Are you sure you installed all dependencies according to the instructions?

> I have many issues with this approach:

The use of `Crypto` was introduced recently in ros/ros_comm#1206. Please consider to provide a pull request to switch to a different implementation.

tfoote, 2019-01-29:
@jwon02 FYI

CdeMills, 2019-01-29:
Hello,

the target is an IoT type platform, so I need to determine "manually" which supplemental dependencies were required w.r.t the basic install.

I found that python-crypto latest release is in 2013; this lib has been superseded by python-cryptodome which implements the same API and add some new function. The latter should thus be the reference implementation for crypto functions.

Regards

Pascal

dirk-thomas, 2019-01-30:
Moved to ros/ros_comm#1599.

closed 2019-01-30 by dirk-thomas