ros/ros #204: mk depends on rospack

moriarty, 2019-02-15:
$ grep -r "shell rospack" ./core/mk/
```
./core/mk/hg_checkout.mk:5:#	include $(shell rospack find mk)/hg_checkout.mk
./core/mk/cmake_stack.mk:22:	$(shell rospack find rosbuild)/bin/package_source.py $(CURDIR)
./core/mk/cmake_stack.mk:34:#include $(shell rospack find mk)/buildtest.mk
./core/mk/git_checkout.mk:5:#	include $(shell rospack find mk)/git_checkout.mk
./core/mk/buildtest.mk:10:rosalldeps = $(shell rospack find rospack)/rosalldeps
./core/mk/buildtest.mk:13:rules = $(foreach d, $(deps), cd $(shell rospack find $(d)) && make &&) true;
./core/mk/buildtest.mk:17:rules_test = $(foreach d, $(deps), cd $(shell rospack find $(d)) && make test &&) true;
./core/mk/buildtest.mk:23:rules_all = $(foreach d, $(deps_all), cd $(shell rospack find $(d)) && make &&) true;
./core/mk/buildtest.mk:27:rules_all_test = $(foreach d, $(deps_all), cd $(shell rospack find $(d)) && make test &&) true;
./core/mk/cmake.mk:49:	awk -f $(shell rospack find mk)/eclipse.awk .project-cmake > .project
./core/mk/cmake.mk:51:	python $(shell rospack find mk)/make_pydev_project.py
./core/mk/cmake.mk:57:include $(shell rospack find mk)/buildtest.mk
./core/mk/bzr_checkout.mk:5:#	include $(shell rospack find mk)/bzr_checkout.mk
```

dirk-thomas, 2019-02-15:
I assume you are trying to point out that the `mk` package lacks a dependency on `rospack` in its `package.xml` file? If yes, please consider to create a pull request to address the problem.

moriarty, 2019-02-16:
Yes sorry I meant to ask a question if it should be a `<build_depend>` or `<build_export_depend>`

I've opened a PR with `build_export_depend`

dirk-thomas, 2019-02-16:
It only needs to be a `build_export_depend` since it is not needed at build time of the `mk` package. Since the manifest uses format 1 that would be a `run_depend`.

dirk-thomas, 2019-02-16:
Thanks for reporting the problem and creating a PR for it.

PR #205 (2019-02-16): [mk] add depend on rospack
Fixes #204
commit 9c92fa4e (2019-02-16): [mk] add depend on rospack (#205)

* [mk] add depend on rospack

Fixes #204

* <depend> instead of <build_export_depend>

Not using package format 2

* build_depend and run_depend

* remove build_depend
closed 2019-02-16 by dirk-thomas