ros/ros #207: rosrun command does not work with nodelets with :: in the name

matt-attack, 2019-02-19:
It is currently impossible to rosrun a nodelet with a :: in the name even though it works through roslaunch.

For example:

This errors

`rosrun nodelet nodelet standalone pcan_driver::PeakCanDriver
type is pcan_driver::PeakCanDriver
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [:] at element [11] is not valid in Graph Resource Name [pcan_driver::PeakCanDriver].  Valid characters are a-z, A-Z, 0-9, / and _.
`

but this works:

`<node pkg="nodelet" type="nodelet"
          name="pcan_device"
          args="standalone pcan_driver::PeakCanDriver"
          machine="$(arg machine)" if="$(arg run_drivers)">
      <rosparam subst_value="true">
        device_address: $(arg can_device)
        baud_rate: 500k
      </rosparam>
</node>`

tfoote, 2019-02-20:
That's an invalid name. If anything that's a bug in the second instance that should error out. Are you sure it's "working?" Please provide enough information to reproduce this.

pjreed, 2019-02-20:
Is there any documentation on what names are valid?  I've been searching around a bit and all I've found is the standard plugin description file documentation: http://wiki.ros.org/pluginlib/PluginDescriptionFile

The relevant part of that says:
> name : The lookup name of the class. Used by the pluginlib tools as an identifier for the plugin. This field is optional in pluginlib 1.9 and higher (ROS Groovy or higher)
> type : The fully qualified class type. 

If name is optional, and nodelets are just plugins that implement the `nodelet::Nodelet` type, this should be a valid nodelet definition:
```xml
<library path="lib/libpcan_driver">
  <class type="pcan_driver::PeakCanDriver"
         base_class_type="nodelet::Nodelet">
    <description>
      Device driver for a Peak CAN adapter.
    </description>
  </class>
</library>
```

It appears to be impossible to launch that nodelet with `rosrun nodelet nodelet standalone ...` because it doesn't have a name and passing the type produces the error that @matt-attack mentioned; on the other hand, a launch file that passes in the type as the second argument works fine.

While investigating, I also noticed that `rosrun nodelet declared_nodelets` does not list that nodelet at all; there are actually several blank lines in its output, which I suspect are other nodelets that do not have names declared.

tfoote, 2019-02-20:
That specific documentation is for pluginlib and for pluginlib to successfully load it does not need the name. However I believe that a name is necessary for nodelet's implementation. (Other packages use pluginlib files.) The error you're getting is that it uses the name as the default node name. And if the name is not set it will use the type. And the type is not a valid ros graph resources.

You can reproduce the error as follows with a nodelet that has a name parameter.
```$ rosrun nodelet nodelet standalone nodelet_tutorial_math/Plus __name:=foo::bar
type is nodelet_tutorial_math/Plus
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [:] at element [3] is not valid in Graph Resource Name [foo::bar].  Valid characters are a-z, A-Z, 0-9, / and _.
Aborted
```

To work around your problem you can just set `__name:=my_node_name` when you rosrun. This is why launching the nodelet is not a problem as it sets the `__name` parameter for all nodes launched.

I would recommend ticketing any nodelets with plugin declarations that don't have names declared following the convention `pkg_name/NodeletName`.

closed 2019-02-20 by tfoote

matt-attack, 2019-02-22:
@tfoote It would be nice if it gave more feedback on what name was invalid so one could guess what is wrong. Should I open up another issue for that?

tfoote, 2019-02-26:
More feedback on that would be great. if you could open a pull request that would be even better. Either way though it should go on the nodelet_core repo though. And please reference back here.