ros/ros #229: How to rosrun a script with the correct python version

Rayman, 2019-10-04:
I'm not sure if this is the correct repo to file the question against, but this is my question.

In the install space, python scripts have their shebang changed to the version specified with `PYTHON_EXECUTABLE`. In the devel workspace this also happens for globally installed scripts. __How can I rosrun a script in the devel space with the correct python version?__

I could change the shebang to `python3` but then it's not backwards compatible. I would like to have this handled automatically.

I've made an example repo to reproduce this problem here:
https://github.com/Rayman/rosrun_bug

```sh
catkin init
catkin config --mkdirs --extend /opt/ros/melodic --cmake-args -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so

cd src
wstool init
curl https://raw.githubusercontent.com/Rayman/rosrun_bug/master/.rosinstall | wstool merge -y -
wstool up

catkin build

source devel/setup.bash 

global-script 
print from /home/user/catkin_ws/src/rosrun_bug/scripts/global-script with 3.6.8

rosrun rosrun_bug rosrun-script 
print from /home/user/catkin_ws/src/rosrun_bug/scripts/rosrun-script with 2.7.15+
```

dirk-thomas, 2019-10-04:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2019-10-04 by dirk-thomas

Rayman, 2019-10-04:
OK, refrased the question here: https://answers.ros.org/question/334621/how-to-create-a-python-script-that-works-with-python3-and-2/