ros/ros #252: Rospy not compatible with python 3.8

kumarvarma1, 2020-04-08:
Rospy works perfectly fine with python 3.6 but in version 3.8 i get the following error when i try to initialize a Ros node

>>> rospy.init_node("name")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 314, in init_node
    logger.info("init_node, name[%s], pid[%s]", resolved_node_name, os.getpid())
  File "/usr/lib/python3.8/logging/__init__.py", line 1434, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.8/logging/__init__.py", line 1565, in _log
    fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
TypeError: findCaller() takes from 1 to 2 positional arguments but 3 were given

dirk-thomas, 2020-04-08:
What exact version of `rospy` are you using and on what platform?

kumarvarma1, 2020-04-08:
I am using Ros Melodic on Ubuntu 18.04

kumarvarma1, 2020-04-08:
I am using Ros melodic on Ubuntu 18.04.I couldn't find my rospy version

dirk-thomas, 2020-04-08:
> What exact version of rospy are you using

`dpkg -l | grep ros-melodic-rospy`

kumarvarma1, 2020-04-08:
1.14.3-0bionic.20191210.231808

> > What exact version of rospy are you using
>1.14.3-0bionic.20191210.231808
> `dpkg -l | grep ros-melodic-rospy`

dirk-thomas, 2020-04-08:
The latest version released in Melodic is 1.14.5. Please make sure to get the latest updates. The problem you describe has been resolved in ros/ros_comm#1838.

closed 2020-04-08 by dirk-thomas