ros/ros #255: TypeError: cannot use a str to initialize an array with typecode 'B'

stertingen, 2020-04-26:
When running a ported version of multimaster_fkie for Python 3, the following error occurs:

```
[ERROR] [1587930573.923334]: Traceback (most recent call last):
  File "<string>", line 109, in main
  File "/home/hermann/turag/programmierung/ros/src/multimaster_fkie/master_discovery_fkie/src/master_discovery_fkie/master_discovery.py", line 581, in __init__
    local_addr = roslib.network.get_local_address()
  File "/opt/ros/melodic/lib/python3.8/site-packages/roslib/network.py", line 139, in get_local_address
    addrs = get_local_addresses()
  File "/opt/ros/melodic/lib/python3.8/site-packages/roslib/network.py", line 187, in get_local_addresses
    buff = array.array('B', '\0' * max_bytes)
TypeError: cannot use a str to initialize an array with typecode 'B'
```

Affected file: https://github.com/ros/ros/blob/noetic-devel/core/roslib/src/roslib/network.py#L197

dirk-thomas, 2020-04-29:
Thanks for the report. Can you please try the proposed patch from #256 and comment if that works for you.

stertingen, 2020-04-30:
> Thanks for the report. Can you please try the proposed patch from #256 and comment if that works for you.

It seems to work now, thanks!

PR #256 (2020-04-30): fix TypeError str / bytes
Fixes #255.
closed 2020-04-30 by dirk-thomas