ros/ros #273: ros.org domain experid

wilcobonestroo, 2020-10-09:
I get a message that the domain name is expired for wiki.ros.org and answers.ros.org. Strangely, discourse.ros.org still works. Is this a local problem (I am located in the Netherlands) or do more people have this issue?
[image]

Frederic98, 2020-10-09:
Same for me (also Netherlands) for the whole ros.org domain, including discourse.ros.org. Maybe that one was cached on your pc?

bwoosley, 2020-10-09:
I am receiving the same thing, located in the Eastern US.

worldpotato, 2020-10-09:
Ah, in germany we have the same issue. Maybe it's still cached in the US or at the VPN provider.

130s, 2020-10-09:
Probably a bit more appropriate repo (and ticket) is https://github.com/osrf/www.ros.org/issues/92 to track the domain issue.

closed 2020-10-09 by dirk-thomas