ros/ros #281: `#!` use python rather than python3 in lots of files

galou, 2021-04-14:
For example the file `core/roslib/src/roslib/launcher.py` uses
```
#! /usr/bin/env python
```

and fails when python is python2 and `python-yaml` or `python-rospkg` are not installed (`python3-yaml` and `python3-rospkg` are installed).

Please note that I didn't have the error myself but report the issue that another person encountered. I'll try to involve him in the issue reporting (for example the run command).

fercuacho, 2021-04-14:
I'm the one who have the problem. 

When i try to launch `roslaunch ur_e_webots ur5e_joint_limited.launch` as explained https://github.com/cyberbotics/webots/tree/released/projects/robots/universal_robots/resources/ros_package/ur_e_webots. 
I got "ImportError: no module named yaml", although I check that I have `python3-yaml` installed. Solved the problem by install the `python-yaml` package. But now the launch process is stopped by "ImportError: no module named rospkg". 

I tried to solve the new problem by in the same way but couldn't because when i tried `sudo apt-get install python-rospkg` for install the `python-rospkg` package, i got:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python-rospkg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-rospkg

E: Package 'python-rospkg' has no installation candidate
```

tfoote, 2021-04-14:
`#!/usr/bin/env python` is used for dual compatible code that can operate with both python2 and python3. Which has been our standard to facilitate the migration from Python 2 to Python 3 for the greater ROS community. Make sure that all your components that you're running in your workspace and in your are using. When you(or the debian build pipeline) invoke the python installer the shebang will be rewritten to your system version of python.

If you're invoking it in a workspace with python2 as the default you need to have all packages and dependencies using python2 as well. There's more info in [PEP 394](https://www.python.org/dev/peps/pep-0394/#recommendation).

If you're consistent with your python environment it works for either version of python. If you don't have python-rospkg available I have to guess that you're on a newer platform and thus must be overriding the python version back to python2. To this end I'd suggest that you ask a question on answers.ros.org with enough information to reproduce your workspace in as reduced a scope as possible to reproduce this problem.

closed 2021-04-14 by tfoote