ros/ros #284: msgpack to rosbag conversion

bkocev, 2021-06-30:
Hi everyone,

we have some Rock (https://github.com/rock-core) data (e.g., GPS, IMU, DVL, Pressure, etc.) serialized and saved into msgpack format (See 1) https://github.com/rock-core/tools-pocolog2msgpack; 2) https://msgpack.org/, and 3) https://github.com/msgpack/msgpack/) and are wondering if we can deserialize the data and save it to a rosbag afterwards so that it can be replayed later using the rosbag functionalities. Any experience with this or hints on how to approach this problem?

Thanks,
Bojan Kocev.

hidmic, 2021-09-10:
@bkocev I don't know if you're familiar with http://answers.ros.org/, but typically we encourage everyone to ask questions over there. That way, it's much more likely you'll get a prompt response from the community and others can benefit from that exchange. Here, it'll get lost once this issue is closed. 

Therefore, I'll go ahead and close this now. Feel free to leave link to your question here for posterity.

closed 2021-09-10 by hidmic

bkocev, 2021-09-10:
@hidmic thanks for pointing that out and reminding me. To whoever might be interested, we took an alternative approach that deploys the rock-ros-bridge, i.e., we do not use the intermediate format msgpack. We load the pocolog data in Rock and forward it to ROS using the above-mentioned bridge.