ros/ros #286: rosbash need to be updated for newer version of fish shell

NextAlone, 2021-08-03:


hidmic, 2021-09-10:
@NextAlone could you be bit more specific? are you willing to contribute?

NextAlone, 2021-09-23:
> @NextAlone could you be bit more specific? are you willing to contribute?

Such as `rostopic pub /vel_cmd geometry/Twist`
It was parsed wrongly with "fish shell string utility with new versions"

Later maybe have some free time to contribute

NextAlone, 2021-12-07:
rostopic pub /rosout rosgraph_msgs/Log  (press tab)

(complete as follows)
[image]

it should be what follows in zsh 
[image]

hidmic, 2021-12-07:
> Later maybe have some free time to contribute

That would be most appreciated.

DragonflyRobotics, 2022-03-13:
The output looks fine to me in bash. I am also running ROS Noetic. I checked on both zsh and bash. Look at the following screenshots:
[image]

[image]

I am not sure if this has been patched already.

NextAlone, 2022-03-15:
> The output looks fine to me in bash. I am also running ROS Noetic. I checked on both zsh and bash. Look at the following screenshots:
> [image]
> 
> [image]
> 
> I am not sure if this has been patched already. 

fish shell will parse the yaml into slices，and regard it as completions，it is not a complete one

DragonflyRobotics, 2022-03-15:
@NextAlone, I am not sure I completely understood what you just said. Do you mean that when we try to autocomplete, we should see the parsed yaml with the \ instead of what you see in my image?

NextAlone, 2022-03-16:
@DragonflyRobotics yes, it maybe not support multi line completion, and,  if there any possibility to convert it to json?