ros/ros #287: Can't install ros-noetic-roslib and ros-noetic-rosconsole in aur

dannkunt, 2021-08-15:
Manjaro GNU/Linux with 5.13.8-1 kernel

logs for ros-noetic-roslib

```
Сборка ros-noetic-roslib...
==> Сборка пакета ros-noetic-roslib 1.15.8-1 (Вс 15 авг 2021 12:03:33)
==> Проверка зависимостей для запуска...
==> Проверка зависимостей для сборки...
==> Получение исходных файлов...
  -> Найден ros-noetic-roslib-1.15.8.tar.gz
==> Проверка файлов source с использованием sha256sums...
    ros-noetic-roslib-1.15.8.tar.gz ... Готово
==> Удаление директории '$srcdir/'...
==> Распаковка исходных файлов...
  -> Распаковка 'ros-noetic-roslib-1.15.8.tar.gz' с помощью bsdtar
==> Удаление директории '$pkgdir/'...
==> Запускается build()...
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /var/tmp/pamac-build-dannkunt/ros-noetic-roslib/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.9.6", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.9/site-packages/em.py  
-- Using empy: /usr/lib/python3.9/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /var/tmp/pamac-build-dannkunt/ros-noetic-roslib/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.5") found components: Interpreter 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-3.9
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Found Boost: /usr/lib64/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: thread 
-- Installing devel-space wrapper /var/tmp/pamac-build-dannkunt/ros-noetic-roslib/src/ros-1.15.8/core/roslib/scripts/gendeps to /var/tmp/pamac-build-dannkunt/ros-noetic-roslib/src/build/devel/lib/roslib
-- Found Boost: /usr/lib64/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: filesystem thread 
-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/pamac-build-dannkunt/ros-noetic-roslib/src/build
make[2]: *** Нет правила для сборки цели «/usr/lib/libboost_filesystem.so.1.75.0», требуемой для «devel/lib/libroslib.so».  Останов.
make[2]: *** Ожидание завершения заданий…
[ 50%] Building CXX object CMakeFiles/roslib.dir/src/package.cpp.o
make[1]: *** [CMakeFiles/Makefile2:275: CMakeFiles/roslib.dir/all] Ошибка 2
make: *** [Makefile:136: all] Ошибка 2
==> ОШИБКА: Произошел сбой в build().
    Прерывание...
```

logs for ros-noetic-rosconsole

```
Сборка ros-noetic-rosconsole...
==> Сборка пакета ros-noetic-rosconsole 1.14.3-2 (Вс 15 авг 2021 12:05:33)
==> Проверка зависимостей для запуска...
==> Проверка зависимостей для сборки...
==> Получение исходных файлов...
  -> Найден ros-noetic-rosconsole-1.14.3.tar.gz
==> Проверка файлов source с использованием sha256sums...
    ros-noetic-rosconsole-1.14.3.tar.gz ... Готово
==> Удаление директории '$srcdir/'...
==> Распаковка исходных файлов...
  -> Распаковка 'ros-noetic-rosconsole-1.14.3.tar.gz' с помощью bsdtar
==> Удаление директории '$pkgdir/'...
==> Запускается build()...
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /var/tmp/pamac-build-dannkunt/ros-noetic-rosconsole/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.9.6", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.9/site-packages/em.py  
-- Using empy: /usr/lib/python3.9/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /var/tmp/pamac-build-dannkunt/ros-noetic-rosconsole/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.5") found components: Interpreter 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-3.9
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Found Boost: /usr/lib64/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: regex system thread 
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- rosconsole backend: glog
-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/pamac-build-dannkunt/ros-noetic-rosconsole/src/build
[ 16%] Building CXX object CMakeFiles/rosconsole_backend_interface.dir/src/rosconsole/rosconsole_backend.cpp.o
[ 33%] Linking CXX shared library devel/lib/librosconsole_backend_interface.so
[ 33%] Built target rosconsole_backend_interface
[ 50%] Building CXX object CMakeFiles/rosconsole_glog.dir/src/rosconsole/impl/rosconsole_glog.cpp.o
[ 66%] Linking CXX shared library devel/lib/librosconsole_glog.so
[ 66%] Built target rosconsole_glog
make[2]: *** Нет правила для сборки цели «/usr/lib/libboost_date_time.so.1.75.0», требуемой для «devel/lib/librosconsole.so».  Останов.
make[2]: *** Ожидание завершения заданий…
[ 83%] Building CXX object CMakeFiles/rosconsole.dir/src/rosconsole/rosconsole.cpp.o
make[1]: *** [CMakeFiles/Makefile2:653: CMakeFiles/rosconsole.dir/all] Ошибка 2
make: *** [Makefile:136: all] Ошибка 2
==> ОШИБКА: Произошел сбой в build().
    Прерывание...
```

hidmic, 2021-09-10:
@dannkunt I don't speak Russian, but if Google Translate is that far off I'm inclined to think this is an issue with Boost in AUR. Check CMake config files in your Boost installation. If everything looks fine, I'd rebuild with fully verbose logging to see what CMake is doing.

dannkunt, 2021-09-11:
[BoostConfig.csv](https://github.com/ros/ros/files/7148362/BoostConfig.csv)
[BoostConfigVersion.csv](https://github.com/ros/ros/files/7148363/BoostConfigVersion.csv)

hidmic, 2021-09-14:
Nothing caught me eye. You'll have to dig a bit deeper @dannkunt.

closed 2025-05-31 by sloretz