ros/ros #288: ROS Noetic Windows 10 runtime 'Debug Error'

cimbar, 2021-08-19:
Hi,
ROS Noetic on Windows 10 fails whenever I try to execute any function with ros namespace. You can find the error dialogue below. It shows `Debug Error!` and `abort() has been called`. I don't face with this issue on python, it is produced only in c++ nodes. I have met with this issue when I called any instruction below, I mean single instruction with ros namespace enough to produce this error. The ROS built version is ros-noetic-desktop_full v20201214.0.0.2104301300.

 ` ros::init(remappings, "sadad");` or
  `ros::NodeHandle nh;` or
  `ros::Rate r(20);` or
  `ros::NodeHandle nh_private("~");` or
  `ros::Duration(1).sleep();`  gives this error: 

[image]

hidmic, 2021-09-10:
@cimbar we'll need a bit more information on your platform (including package versions) plus a small example to reproduce.

cimbar, 2021-09-11:
I find out the problem occurs when debug mode enabled in CMakeLists file. When I removed the `set(CMAKE_BUILD_TYPE Debug)`, the issue is solved.

closed 2025-05-31 by sloretz