ros/ros #289: Message getting stuck

saadnec, 2021-09-17:
I am having an issue while using multiple uavs in Gazebo using ros. I am currently using 3 UAVs, sometimes the 3rd UAV doesn't take off, so I have to close the gazebo and relaunch it again and then all 3 takes off and the program proceeds. What I understand is that the message is not passed correctly. The message gets stuck in the buffer. Is there a way to empty the buffer ?

osuairt, 2021-09-17:
You've provided next to no details for anyone to be able to troubleshoot that problem for you.

hidmic, 2021-09-17:
I echo @osuairt. Moreover, this is neither a feature nor a bug report, but a question. 

I'd encourage you to make questions at http://answers.ros.org. It's makes it way more likely you'll get a response from the broader community, and other can benefit from that exchange in the future. Here, it'll get lost once the issues closes. Therefore, I'll close this ticket now.

closed 2021-09-17 by hidmic