ros/ros #292: why can't I run catkin_make in ROS windows?

ZA0068, 2021-12-26:
I have installed ROS Noetic on my system and followed the tutorial completely and carefully on those steps.

Everything works fine and I have Visual Studio 2022. I tested roscpp and stuff and they worked like a charm. I also installed ROS2 Foxy and it also works well on my Visual Studio 2022.

But when I create Catkin workspace on any folder, I get an error that I can't fix.

I deleted and remade my workspace but I get the same error.

This is the entire error here:
'''
Base path: C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws
Source space: C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\src
Build space: C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\build
Devel space: C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\devel
Install space: C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\install
Copying file from "c:\opt\ros\noetic\x64\share\catkin\cmake\toplevel.cmake" to "C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\src\CMakeLists.txt"
Running command: "cmake C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\install -G Ninja" in "C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\build"

-- The C compiler identification is MSVC 19.30.30706.0
-- The CXX compiler identification is MSVC 19.30.30706.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;c:\opt\ros\noetic\x64
-- This workspace overlays: c:/opt/ros/noetic/x64
-- Found PythonInterp: C:/dev/MySys/mingw64/bin/python3.exe (found suitable version "3.9.7", minimum required is "3")
-- Using PYTHON_EXECUTABLE: C:/dev/MySys/mingw64/bin/python3.exe
-- Using default Python package layout
-- Found PY_em: c:\opt\ros\noetic\x64\lib\site-packages\em.py
-- Using empy: c:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/Users/zaina/anaconda3/Scripts/nosetests-3.9
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
Python path configuration:
PYTHONHOME = 'C:\dev\MySys\mingw64\bin'
PYTHONPATH = 'c:\opt\ros\noetic\x64\lib/site-packages;c:\opt\ros\noetic\x64\Lib\site-packages'
program name = 'C:\dev\MySys\mingw64\bin\python3.exe'
isolated = 0
environment = 1
user site = 1
import site = 1
sys._base_executable = 'C:\dev\MySys\mingw64\bin\python3.exe'
sys.base_prefix = 'C:\dev\MySys\mingw64\bin'
sys.base_exec_prefix = 'C:\dev\MySys\mingw64\bin'
sys.platlibdir = 'lib'
sys.executable = 'C:\dev\MySys\mingw64\bin\python3.exe'
sys.prefix = 'C:\dev\MySys\mingw64\bin'
sys.exec_prefix = 'C:\dev\MySys\mingw64\bin'
sys.path = [
'c:\opt\ros\noetic\x64\lib/site-packages',
'c:\opt\ros\noetic\x64\Lib\site-packages',
'C:\dev\MySys\mingw64\bin\lib\python39.zip',
'C:\dev\MySys\mingw64\bin\lib\python3.9',
'C:\dev\MySys\mingw64\bin\lib\python3.9\lib-dynload',
'',
]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x000031c4 (most recent call first):

The syntax of the command is incorrect.
Traceback (most recent call last):
File "C:\Users\zaina\OneDrive\Dokumenter\ros_ws\noetic_ros_1\catkin_ws\build\catkin_generated\generate_cached_setup.py", line 22, in
code = generate_environment_script('C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/devel/env.bat')
File "c:\opt\ros\noetic\x64\lib\site-packages\catkin\environment_cache.py", line 62, in generate_environment_script
output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
File "C:\dev\MySys\mingw64\lib\python3.9\subprocess.py", line 424, in check_output
return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
File "C:\dev\MySys\mingw64\lib\python3.9\subprocess.py", line 528, in run
raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/devel/env.bat', 'C:\dev\MySys\mingw64\bin\python3.exe', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 255.
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/safe_execute_process.cmake:11 (message):
execute_process(C:/dev/MySys/mingw64/bin/python3.exe
"C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/build/catkin_generated/generate_cached_setup.py")
returned error code 1
Call Stack (most recent call first):
C:/opt/ros/noetic/x64/share/catkin/cmake/all.cmake:208 (safe_execute_process)
C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:20 (include)
CMakeLists.txt:58 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/zaina/OneDrive/Dokumenter/ros_ws/noetic_ros_1/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
'''

I need help, how do I fix this issue?

ZA0068, 2021-12-28:
is someone there?

samuellimabraz, 2023-08-23:
I installed ros-noetic on windows and got the same problem. I was able to solve it by specifying the python exucatable: `catkin_make -DPYTHON_EXECUTABLE=<python.exe path>`