ros/ros #295: rosbash: rosrun does not auto-complete executable (node) if workspace is in hidden folder

svddries, 2022-01-07:
Hey there!

I found that if a ROS / catkin workspace is located in a hidden folder, `rosrun` *will* complete the package name, but not the executable. A minimal example (Ubuntu 20.04, ROS Noetic):

```bash
source /opt/ros/noetic/setup.bash

# Create a workspace in a hidden directory
mkdir -p ~/.catkin_ws/src
cd ~/.catkin_ws/

# Get some package with executables
git clone https://github.com/ros/catkin_tutorials /tmp/catkin_tutorials
cp /tmp/catkin_tutorials/create_package_pubsub/catkin_ws/src/beginner_tutorials src -r

catkin_make
source devel/setup.bash

# Completing the package works:
rosrun beginner<TAB>

# But auto-complete on the executable doesn't
rosrun beginner_tutorials <TAB>
```

If you follow these steps with `~/catkin_ws` instead of `~/.catkin_ws`, completion on executables *does* work as expected.

This case may seem a bit contrived (who on earth puts the workspace in a hidden directory?!), but we encountered the issue when combining ROS with Conda: in our case, we put the Conda environment in a `.env` directory such that it's nicely hidden from the user, but it caused `rosrun` to not work correctly for Conda-installed ROS packages. Our simple fix for now is simply renaming the `.env` folder to something non-hidden.

I think I also found the cause of the issue. See [line 261 in `rosbash`](https://github.com/ros/ros/blob/noetic-devel/tools/rosbash/rosbash#L261): the `! -regex .*/[.].*` option removes all paths that contain a `.`. I'm not sure though why this filter is added, so I wasn't sure how this can be fixed.

Thanks in advance!

FSund, 2022-08-22:
I'm having the same issue. I'm running ROS in a distrobox, and have put my home folder in `$HOME/.distrobox/home/noetic`, so I presume it's the dot that's causing the issue.

FSund, 2022-08-22:
I just tested removing all 7 instances of `! -regex ".*/[.][^./].*"` and the one instance of `! -regex .*/[.].*` from [rosbash](https://github.com/ros/ros/blob/noetic-devel/tools/rosbash/rosbash), and now autocomplete works fine inside the dot-prefixed folder.

FSund, 2022-08-22:
I had a gander through the history of rosbash, and it seems like the dot-folder exclusion rule has been there since the start. [Here](https://github.com/ros/ros/blob/8a1a03f015e0f374c1dd41eaefde5ebe206fa009/tools/rosbash/rosbash#L160) is the earliest version of the file, and it still has the regex.

The regexes were modified in https://github.com/ros/ros/commit/06a3597afed12f6c36b608b24964c24fa41e4199 to not remove paths `./` or `../`.

What would the effect of removing the regex be? Or what effect is it supposed to have?

closed 2025-05-31 by sloretz