ros/ros #302: ModuleNotFoundError: No module named 'roslib'

is-HUKai, 2022-04-02:
[rosbuild] Building package ORB_SLAM2_PointMap_SegNetM
[rosbuild] Error from syntax check of ORB_SLAM2_PointMap_SegNetM/manifest.xml
Traceback (most recent call last):
File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'roslib'
CMake Error at /opt/ros/melodic/share/ros/core/rosbuild/private.cmake:77 (message):
[rosbuild] Syntax check of ORB_SLAM2_PointMap_SegNetM/manifest.xml failed;
aborting
Call Stack (most recent call first):
/opt/ros/melodic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
CMakeLists.txt:4 (rosbuild_init)

yiliaLuo, 2022-06-15:
hi,did u solve the problem?

i met the same problem while executing the command "cmake .. -DCMAKE_BUILD_TYPE=Release" in the path of "catkin_ws/src/DS-SLAM/Examples/ROS/ORB_SLAM2_PointMap_SegNetM/build".

and the terminal showed that:

`[rosbuild] Building package ORB_SLAM2_PointMap_SegNetM
[rosbuild] Error from syntax check of ORB_SLAM2_PointMap_SegNetM/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ImportError: No module named rospkg
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM2_PointMap_SegNetM/manifest.xml failed;
  aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)`

yiliaLuo, 2022-06-15:
it seems to be solved by install a higher version of python. since i use the ubuntu 20.04, i need to install python3.

reference : https://answers.ros.org/question/245967/importerror-no-module-named-rospkg-python3-solved/

i execute `sudo apt install python-is-python3` in the terminal , and you can check the version of python by `python --version`.

closed 2025-05-31 by sloretz