ros/ros #305: ROS networks

avillalva, 2022-06-19:
hi,

I have a ROS publisher on a raspberry pi 4 and a ROS subscriber on a home PC. I have an odd quirk where the PC based subscriber does not initially hear published messages when the pi is on wireless. They are both on the same private network 192.168.0.0/24. I can confirm that the wireless on the pi is working because I am SSH onto it, and once I plug in the Ethernet cable to the pi the whole system (publisher / subscriber) begins to work. The odd thing is that I can then unplug the ethernet cable from the pi and it continues to work over wireless. But it does need that kick start via ethernet.

I can ping the PC from the pi and I can ping the pi from the PC over wireless. Any thoughts on how to resolve? I need it to work over wireless without the intermediate step of plugging in the ethernet cable.

tfa27, 2022-06-19:
is your ethernet port configured in your  `/etc/network/interfaces.d/interfaces` file? Can you echo topics on the pi before doing the ethernet port trick?

avillalva, 2022-06-21:
hi,

Yes, the ethernet port is configured and working correctly. I can echo topics published by the same device e.g. pi published topics can be echoed successfully on the pi and vm published topics can be echoed successfully on the vm. 

Following that trail, I ran a packet capture on the pi and found that traffic is published to the loopback interface - Not to the eth0 or the wlan0 interfaces. This explains the lack of visibility between machines. 

route tables show the default route as wlan0 with eth0 as a backup. The loopback interface is not represented at all in the route tables so I'm at a loss as to why its publishing to itself only?

closed 2025-05-31 by sloretz