ros/ros #307: The repository 'http://packages.ros.org/ros/ubuntu jammy Release' does not have a Release file.

programmeddeath1, 2022-09-26:
I am getting this error on trying to install ros on ubuntu 22.04

[image]

Is ubuntu 22.04 not supported? I tried building from source too, but am getting a failure due to a dependency from rosedep.
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
diagnostic_common_diagnostics: [hddtemp] defined as "not available" for OS version [*]
```

Any help would be appreciated! Thank you

erikbgithub, 2022-09-26:
I get the same error with Debian Bullseye. I also found a similar question on the [community help forums](https://answers.ros.org/question/406898/can-i-install-ros-melodic-in-ubuntu-2204-lts/).

I'll make an account there and link this issue there as well.

gavanderhoorn, 2022-09-26:
There are no supported ROS 1 versions for Ubuntu `22.04`. `packages.ros.org` does not host any `.deb`s for that distribution, hence no release file, hence the error message.

I've answered the question linked by @erikbgithub with a similar post.

closed 2025-05-31 by sloretz