ros/ros #310: A problem in topic

bromiao123, 2023-02-07:
When I running this program

sometimes it causes this problem:
{3: 'Traceback (most recent call last):
  File "3.py", line 62, in <module>
    talker()
  File "3.py", line 53, in talker
    
pub.publish(String("hello world {}".format(count)))
 File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1041, in publish
    raise ROSException("publish() to a closed topic")
rospy.exceptions.ROSException: publish() to a closed topic
'}

bromiao123, 2023-02-07:
the code is following:

def talker():
    pub = rospy.Publisher("chatter", String, queue_size=1)
    rospy.init_node("testnode3", anonymous=True)
    freq = rospy.get_param("~frequency", 10)
    rate = rospy.Rate(freq)

    count = 0
    while not rospy.is_shutdown():
        pub.publish(String("hello world {}".format(count)))
        count += 1
        rate.sleep()

tfoote, 2023-05-15:
Please ask questions on http://answers.ros.org according to our support policy: http://wiki.ros.org/Support We try to keep the issue tracker reserved for bugs and feature developments.

closed 2023-05-15 by tfoote