ros/ros #314: RLException : error loading <rosparam> tag

Frexicane, 2023-06-04:
Hello Guys! 

So I keep trying to launch the navigation files with roslaunch however I get this message only with this navigation package. I don't see a clear answer on why I'm getting this error. The only problem I could see is on line 46 where there are empty "" but even when I add any of the costmap_converter files, I still get the same error.
[image]
[image]
[image]
 Note* I am new to ROS and learning. I will provide images of the YAML file and the error below.

Keythings to note, I am using the PuppyPi Pro from HiWonder if anyone has experience with any of their products*

the error states : RLException : error loading <rosparam> tag : file /home/pi/puppy_pi/src/puppy_navigation/config/teb_local_planner_params.yaml contains invalid YAML: unacceptable character #x0080: special characters are not allowed in "<unicode string>", position 384. XML is <rosparam command= "load" file="$(find puppy_navigation)/config/teb_local_planner_params.yaml"/> The traceback for the exception was written to the log file

insomniacgator, 2024-04-19:
Did you ever fix this?

Frexicane, 2024-05-21:
> Did you ever fix this?

I wish