ros/ros #316: Issue getting rosbag

ashwinjoseph95, 2023-06-20:
Hey Team,
I was following the tutorial on this link: [](http://wiki.ros.org/rosbag/Tutorials/reading%20msgs%20from%20a%20bag%20file) and was trying to get the bag file based on the command shown there:
`wget https://open-source-webviz-ui.s3.amazonaws.com/demo.bag`
But end up facing an issue saying:
```
--2023-06-20 20:50:20--  https://open-source-webviz-ui.s3.amazonaws.com/demo.bag
Resolving open-source-webviz-ui.s3.amazonaws.com (open-source-webviz-ui.s3.amazonaws.com)... 54.231.164.65, 3.5.25.228, 52.217.80.44, ...
Connecting to open-source-webviz-ui.s3.amazonaws.com (open-source-webviz-ui.s3.amazonaws.com)|54.231.164.65|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-06-20 20:50:21 ERROR 404: Not Found.

```
I understand the ROS community may not be directly responsible for maintaining this bag file but as the link does not seem to work (I find issues raised on the same that date back at least for over 6 months ), please either update with another example or provide an alternative way to get the bag file please.

askman92, 2024-02-24:
Similar issue but I get a 403 forbidden error

liuxk99, 2024-10-31:
I get a 403 error too.
```
$ wget https://open-source-webviz-ui.s3.amazonaws.com/demo.bag
--2024-10-31 10:54:55--  https://open-source-webviz-ui.s3.amazonaws.com/demo.bag
Resolving open-source-webviz-ui.s3.amazonaws.com (open-source-webviz-ui.s3.amazonaws.com)... 16.15.217.186, 3.5.22.240, 52.217.231.49, ...
Connecting to open-source-webviz-ui.s3.amazonaws.com (open-source-webviz-ui.s3.amazonaws.com)|16.15.217.186|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2024-10-31 10:54:57 ERROR 403: Forbidden.

joshuayanggithub, 2024-12-22:
It seems this section has been partly cut from the tutorial link, but I was able to fetch the correct demo.bag file through [Wayback Machine](https://web.archive.org/web/20200910023556/https://open-source-webviz-ui.s3.amazonaws.com/demo.bag) thankfully (It has been archived in 2020)

closed 2025-05-31 by sloretz